kreso-t/cncjs-kt-ext #1: Question about running on console vs in a macro

TSwimson, 2019-05-28:
Hey thanks for making this! Looking forward to trying it out. I was trying to use it last night by typing (#autolevel) into the console and the autolevel kept crashing. After poking around the code I noticed that context was undefined in the start function. It looks to me like when you send the command from the console context is not set but if you send the command from a macro then it is set. Any idea why this is? Is that a bug in CNCjs?

kreso-t, 2019-05-28:
Hey, I've been using it only from macro recently, now I checked and also for me does not work from console. I think it did work in the beginning (I upgraded my CNCjs in the meanwhile). Anyway it is much more convenient use it as macro, so I will just update the readme file about this limitation for now.

closed 2019-05-28 by kreso-t

TSwimson, 2019-05-28:
Ok thanks!