kreso-t/cncjs-kt-ext #2: undefined socket when running "npm start"

codehero, 2019-07-28:
Get the following error message when starting.
I have set the port option to the correct serial port as well.

```
Connected to ws://localhost:8000?token=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpZCI6IiIsIm5hbWUiOiJjbmNqcy1hdXRvbGV2ZWwiLCJpYXQiOjE1NjQzMjQ3NzcsImV4cCI6MTU2NjkxNjc3N30.ed3RVJ13fwN-h3MHlWYYl8_IfgZN82m68-UuQtAx7YQ
/home/pi/src/cncjs-kt-ext/autolevel.js:21
    socket.on('gcode:load', (file, gc) => {
          ^

TypeError: Cannot read property 'on' of undefined
    at new Autolevel (/home/pi/src/cncjs-kt-ext/autolevel.js:21:11)
    at callback (/home/pi/src/cncjs-kt-ext/index.js:99:19)
    at Socket.<anonymous> (/home/pi/src/cncjs-kt-ext/index.js:94:3)
    at Socket.Emitter.emit (/home/pi/src/cncjs-kt-ext/node_modules/component-emitter/index.js:133:20)
    at Socket.onevent (/home/pi/src/cncjs-kt-ext/node_modules/socket.io-client/lib/socket.js:278:10)
    at Socket.onpacket (/home/pi/src/cncjs-kt-ext/node_modules/socket.io-client/lib/socket.js:236:12)
    at Manager.<anonymous> (/home/pi/src/cncjs-kt-ext/node_modules/component-bind/index.js:21:15)
    at Manager.Emitter.emit (/home/pi/src/cncjs-kt-ext/node_modules/component-emitter/index.js:133:20)
    at Manager.ondecoded (/home/pi/src/cncjs-kt-ext/node_modules/socket.io-client/lib/manager.js:345:8)
    at Decoder.<anonymous> (/home/pi/src/cncjs-kt-ext/node_modules/component-bind/index.js:21:15)
```

kreso-t, 2019-08-11:
Hi,

based on my tests "npm start" seem to remove the arguments starting with "-"

Can you try run the program and set the options in the following way:
node . --port {port} 

or modify the package.json start script and include options there, i.e.

package.json:
```
  "scripts": {
     "start": "node .  --port {port}"
   }
```

BR,
Kresimir

codehero, 2019-08-12:
Running

`node . --port {port}`

seems to work for me. Thanks.

closed 2019-08-12 by codehero

Franckapik, 2020-11-07:
Hi,
Is there a way to avoid this error when using pm2 with port /dev/ttyS0 ? 
The app works when i use "node . --port /dev/ttyS0" but not when i use pm2...

opit7, 2022-04-27:
@Franckapik the following works for me.
`module.exports = {
  apps : [
      {
        name: "cncjs-kt-autolevel",
        script: ".",
        args: "--port /dev/ttyS0"
      }
  ]
}`