kreso-t/cncjs-kt-ext #4: Autoleveling Error

chrishdx, 2019-09-15:
When I use the Autoleveling function, he changes the gcode so that he mows deep into the board. I have already installed a 100 nf CAP.

Gcode before autoleveling

Axis Min Max dimension
X 0.000 mm 91.486 mm 91.486 mm
Y 0.000 mm 65.814 mm 65.814 mm
Z -0,100 mm 20,700 mm 20,800 mm

after carlevel

Axis Min Max dimension
X 0.000 mm 91.486 mm 91.486 mm
Y 0.000 mm 65.814 mm 65.814 mm
Z -1,345 mm 20,700 mm 22,045 mm

Output Commandline

STEP: 25 mm HEIGHT: 0.7 mm FEED: 100
WCO: {x: -1.8136, y: -1.2275, z: -0.014300000000000007}
probed 1/12> 0.000 0.000 0.000
probed 2/12> 1.201 0.000 0.006
probed 3/12> 2.401 0.000 0.006
probed 4/12> 3.602 0.000 -0.003
probed 5/12> 0.000 1.296 0.005
probed 6/12> 1.201 1.296 0.010
probed 7/12> 2.401 1.296 0.009
probed 8/12> 3.602 1.296 -0.005
probed 9/12> 0.000 2,591 0.005
probed 10/12> 1.201 2.591 0.012
probed 11/12> 2.401 2.591 0.008
probed 12/12> 3.602 2.591 -0.007
apply leveling
Leveling applied

Can someone help me what is not going right here I use as a unit mm (G21)

kreso-t, 2019-09-15:
Would it be possible to attach the original g-code file?

chrishdx, 2019-09-15:
Hi thanks for your quick response

chrishdx, 2019-09-15:
https://1drv.ms/u/s!Ah9Na_8sr-vQhK9JRMzPw4FRpPnIQw?e=r9n6mj

kreso-t, 2019-09-15:
Thx,

I tried with your file but my output is different and end result looks OK:

before leveling:
Axis | Min | Max | Dimension
X | 0.000 mm | 81.761 mm | 81.761 mm
Y | 0.000 mm | 61.926 mm | 61.926 mm
Z | -0.200 mm | 20.700 mm | 20.900 mm

after leveling:
Axis	Min	Max	Dimension
X	0.000 mm	81.761 mm	81.761 mm
Y	0.000 mm	61.926 mm	61.926 mm
Z	-0.324 mm	20.762 mm	21.086 mm

so it is only 0.124 mm deeper  after leveling

my cmd line output:

STEP: 25 mm HEIGHT:0.7 mm FEED:100
WCO: { x: -259.878, y: -105.08, z: -104.13000000000001 }
probed 1/12> 0.000 0.000 0.062
probed 2/12> 27.255 0.000 0.032
probed 3/12> 54.508 0.000 0.110
probed 4/12> 81.760 0.000 0.162
probed 5/12> 0.000 30.962 -0.140
probed 6/12> 27.255 30.962 -0.060
probed 7/12> 54.508 30.962 -0.008
probed 8/12> 81.760 30.962 -0.120
probed 9/12> 0.000 61.925 -0.010
probed 10/12> 27.255 61.925 0.150
probed 11/12> 54.508 61.925 0.192
probed 12/12> 81.760 61.925 0.062

it looks as if in your case the coordinates in the NC file were actually in inch
i.e. point 2
probed 2/12> 1.201 0.000 0.006 (from your example - looks like the approx. same XY coordinate as below but in here inch) 
probed 2/12> 27.255 0.000 0.0 (from my example) 

BR,
K.

chrishdx, 2019-09-16:
Mhh very strange do you use mm or inches? Do you have an idea why this
could be?

chrishdx, 2019-09-16:
Have found the error was due to my GRBL configuration $ 13 = 0 (Report inches, boolean) it was Set 1 by derfault. I have changed to 0 now he reports in mm not inches.  However, a thousand thanks for your help

closed 2019-09-16 by chrishdx