kreso-t/cncjs-kt-ext #5: Auto-start extention

brettlx, 2019-09-17:
Hi. Thanks for your work on this. This is not an issue, but rather a question.
Have you set your environment up to auto start this application when the host starts?
If so, how did you do it. I've looked at using /etc/rc.local with a delay start to give CNC.js time to start up but not sure if this is the best way.

TimW55, 2019-09-23:
As cncjs takes some time to start I found the easiest way to start the application was to added a command to cncjs which I run manually when I first power up the router:
``
cd ~/cncjs-kt-ext && node . --port /dev/ttyUSB0 </dev/null &>/dev/null &
``

brettlx, 2019-09-23:
Thanks for responding Tim. The autorun with rc.local works okay (I have put in a 60 second delay), but your solution is more sensible. Don't have to run it if not needing to use it.

closed 2019-09-23 by brettlx