kreso-t/cncjs-kt-ext #8: Working with the app ?

xseignard, 2019-11-15:
Hi,
I'm trying to use your extension, but when I run the macro from the app, nothing happens.

Do I miss something ?

Running cncjs 1.9.20 and the master branch of your repo

Regards

greg-wood, 2019-11-27:
Twelve days and no response?  This is not encouraging. I know little to nothing about this as I am currently tying to set it up myself. I did however notice you have to have your g-code loaded as that is where it gets the boundaries.    Did you have your pcb g-code loaded?

kreso-t, 2019-11-27:
Hi,

@xseignard: I would need more specific details about the steps you did, your configuration, how did you run the extension,  output you get from the console, etc.  in order to give you some meaningful advice. Could it be that your machine is not on the port /dev/ttyACM0 and you did not provide the actual port in the arguments (--port)?

BR,
K.

kreso-t, 2019-11-27:
> Twelve days and no response? This is not encouraging. I know little to nothing about this as I am currently tying to set it up myself. I did however notice you have to have your g-code loaded as that is where it gets the boundaries. Did you have your pcb g-code loaded?

@greg-wood: sorry for the late reply, did not notice the post until now. However it's not my duty to provide timely support for this code and also it is not my duty to encourage you. It is just my good will to do so, and I am starting to have less of it, seeing your comment...

greg-wood, 2019-11-27:
Yeap, the attitude matches the experience/results. It’s a shame though, i got mine working and it’s a great app.

For the other poster it probably is the port. There is no open failure error message. Screen just hangs.

I pretty much supplied all the options; except secret and it came right up. Im using the desktop app version of cncjs and had to select the “open in browser ” choice in one of the cncjs pull down menus. After it opened i could get the port number.   It seems to change everything you restart cncjs.

kreso-t, 2019-12-15:
Closing this, assuming the problem was with the --port setting

closed 2019-12-15 by kreso-t