kreso-t/cncjs-kt-ext #10: Crash after leveling

codehero, 2019-12-14:
Leveling proceeds fine, but at the very end this has happened to me several times already and it seems random:

probed 363/364> 124.558 65.748 -0.020
probed 364/364> 129.748 65.748 -0.035
apply leveling
**Leveling applied
zlib.js:623
      var newReq = self._handle.write(flushFlag,
                               ^

TypeError: Cannot read property 'write' of null
    at Zlib.callback (zlib.js:623:32)**

kreso-t, 2019-12-15:
Hi,
Which version of cncjs are you using? Does the gcode modified by autolevel load after the leveling process?
Can you try with the most recent version of the extension,  there are some additional grbl console messages/gcode comments written during the process, can you post them when this happens again.
Thx

codehero, 2019-12-15:
I am using cncjs 1.9.20
I am not sure if the gcode loads but I see badly leveled boards made.

Can you instruct me on which git revision to try?

Thanks for your help!

kreso-t, 2019-12-15:
when new modified gcode gets loaded by the cncjs-kt-ext it adds the '#AL:' prefix to the name i.e. "#AL:<original_gcode_file_name>", so this way you know that compensation has applied 
(the filename is visible i.e. at bottom-left corner of the 3d View in cncjs)

last master version should work i.e. 
`git clone https://github.com/kreso-t/cncjs-kt-ext.git`

codehero, 2019-12-15:
FYI this was my rev e36cebc2b6cd8fd57010264304446b65ddc508fc

I will test and let you know

codehero, 2019-12-18:
OK I tried the latest revision and while the autolevel did not crash, the autolevel mangles my gcode in the xy dimensions and generates bad gcode as the following:

(original gcode attached)

[image]

[bracket2.txt](https://github.com/kreso-t/cncjs-kt-ext/files/3977272/bracket2.txt)

kreso-t, 2019-12-18:
I see that you use G2/G3 circular interpolation moves in gcode, these are not supported by this extension as G2/G3 moves are normally not generated by the PCB milling generators (I use [FlatCAM ]( http://flatcam.org/) for generating PCB gcode).

closed 2021-03-18 by kreso-t