kreso-t/cncjs-kt-ext #11: Installation

leoheck, 2019-12-25:
Hi, I am following your instructions to install it, to give it a try.

I am receiving this:
[image]

Do you have any idea what I am doing wrong?

leoheck, 2019-12-25:
I am also looking at the help and I have some issues.

The `-s, --secret` parameter does not show a `<value>` filed next to it like in `port` flag, for example.  Is it right? Also, some parameters like `-c, --config <filename>` show that the default is `~/.cncrc` but also show `(default: "")` at the end of the string. Is it right?

```
➜  cncjs-kt-ext git:(master) node . --help
Usage: cncjs-kt-ext -s <secret> -p <port> [options]

Options:
  -V, --version                       output the version number
  -s, --secret                        the secret key stored in the ~/.cncrc file
  -p, --port <port>                   path or name of serial port (default: "")
  -b, --baudrate <baudrate>           baud rate (default: 115200) (default: 0)
  -c, --config <filename>             set the config file (deafult: ~/.cncrc) (default: "")
  --socket-address <address>          socket address or hostname (default: localhost) (default: "")
  --socket-port <port>                socket port (default: 8000) (default: "")
  --controller-type <type>            controller type: Grbl|Smoothie|TinyG (default: Grbl) (default: "")
  --access-token-lifetime <lifetime>  access token lifetime in seconds or a time span string (default: 30d) (default: "")
  -h, --help                          output usage information
```

leoheck, 2019-12-25:
Also, the `-s, --secret` does not work

[image]

leoheck, 2019-12-25:
I figured out that I can use the `CNCJS_SECRET` environment variable. So now, I am receiving the following output:

```
Connection error. Unauthorized user: user.id=, user.name=cncjs-autolevel
```

How do I set these user.id and user.name?

leoheck, 2019-12-25:
Alright, I was able to run it by adding my credentials directly on the `id` and `name` fields on `generateAccessToken` function.

matthgyver, 2019-12-25:
Thank's guy !
Little precision : ID could be found in the ``~/.cncrc`` file

leoheck, 2019-12-26:
@matthgyver Yeap that is right. I am fixing these things on my fork. I was not able to test it yet. I just know that it is running. If I can fix these command-line issues I will make a pull request here.

closed 2020-01-10 by leoheck