kreso-t/cncjs-kt-ext #12: Autolevel more than one time

leoheck, 2020-01-06:
I was trying to use the auto level and my CNCjs yesterday. I am new to this field and tools. So, I was playing a lot with Flatcam + CNCjs + your plugin.

As I could understand, I can't run the auto level twice. It will make things deeper than expected (I guess). 

So, my question is, what happens when we use the auto leveler twice? It will use the originally uploaded gcode, or the previews altered gcode? 

Also, if it uses the previews altered gcode, it will mess the final product or it will keep it fine?

If this is some kind of bug, is it possible to always use the original file for the auto leveler as the input of this plugin instead of using the already altered file?

kreso-t, 2020-01-07:
Hi,

It is supposed to use original gcode file every time you run the autolevel (but you need to keep the plugin process running the whole time - since the original file is stored in the memory only). However I have not use it in this way often. I usually reload the gcode file and then run the autolevel again.  Please let me know if you find any problem with this.

BR,
K.

leoheck, 2020-01-07:
Good, thank you. I will add this info in the readme.

I am running both cncjs and your plugin on a rpi, so both have its own service that starts with the system.

closed 2020-04-09 by leoheck