kreso-t/cncjs-kt-ext #15: (Issue) The isolation routing is too deep.

leoheck, 2020-01-10:
Issue running CNC.

@kreso-t I was not able to make a good PCB, yet. 
Can you assist me with that?

I am using Kicad to create a simple/single-layer board.
Then I am exporting Gerbers in millimeters.

After, I am importing the `B.Cu` layer in Flatcam
And exporting the Gcode with the following settings:
- **Tool Diameter**: 0.127 mm
- **Cut Z**: -0.05 mm
- **Travel Z**: 0.2
The .gcode looks good, especially the `G21` saying units are millimeters. And the Z travel distance is -0.05 mm too.

Then, running CNCJs and CNCJs-Kt-Ext, I am:
1. Importing the .gcode generated with Flatcam.
2. Then, jogging the drill bit to a place that will be set as (x=0, y=0)
3. Then, setting zero the working position. Considering X, Y, and Z the axis. **(Z is still 1 or 2 mm over the board)**
4. Then, running `(# autorun)` macro. With probes. This process works perfectly. The .gcode is updated and the text `#AL` is added on the .gcode name at the bottom of the page.
5. Then,  I HIT play, to start drilling the PCB, the drill bit is going too deep (like -1,5 mm) opening a huge hole instead of just removing the cooper on the surface (it should be - 0.05 mm instead).

What I am doing wrong? Do you have any idea? 
I don't know if step #3 is correct.

kreso-t, 2020-01-10:
The procedure you described seems correct, it should work like so. In step #3 I usually also zero the Z axis (by using Z probe), so that tool is in working position x=0, y=0, z=0 when touching the PCB surface at the origin point. Then I lift it a bit and start autolevel. Not sure if I ever tried it without properly zero-ing the Z axis ... but it should also work if Z is at least approximately zeroed.

Travel Z: 0.2 may be too low unless your PCB is very small and very flat - but this should not cause this problem that drill bit is going to deep.

Before actually running the gcode - it's good to check/review the Z coordinates in the gcode file before autoleveling and after it - if there are big delta in the Z-coordinates then something may went wrong with the process

leoheck, 2020-01-10:
Thank you for your comments :)

> so that tool is in working position x=0, y=0, z=0 when touching the PCB surface...

So, I always zero-ing the Z-axis in the CNCjs.
**But my doubt is, should I move/probe the Z-axis to touch the bed before zero-ing it?** 
If it is true, I am not doing this right.

> Travel Z: 0.2

Oh, you are right! I was thinking 2 mm, but I was using 0.2 instead.

> it's good to check/review the Z coordinates in the gcode file before auto-leveling and after it - if there are big delta in the Z-coordinates then something may went wrong with the process.

How much is this big delta you are talking about?

So, regarding that, what are your thoughts on adding a threshold in `#autolevel` macro to display a warning when this big delta is bigger than expected?

We can still keep the original macros working, like
```
(#autolevel) ------------------> Current macro that uses all default values
(#autolevel Dxx Hxx Fxx) ------> Current macro that uses the default big delta threshold
(#autolevel Dxx Hxx Fxx Txx) --> New macro to set the big delta threshold explicitly
```

leoheck, 2020-01-11:
The best try so far is this. Now, the drill barely touches the bed.
I still don't know how to set up the initial Z. Do you have any tutorial to share?

[image]

After the bed-leveling, these diagonal lines appear, but they are being drawn in the PCB as you can see in the first picture.

[image]

Here, the bed leveling output, the deltas are correct this time, I gues. 

[image]

leoheck, 2020-01-11:
I improved the isolation routing by increasing FlatCam parameters. It happens that my fresa bit is worn l compared to a new one. I will repeat my tests later today.

kreso-t, 2020-01-11:
About the zero-ing: "But my doubt is, should I move/probe the Z-axis to touch the bed before zero-ing it?"
Yes, tool should be touching the surface when Z=0.

closed 2020-04-09 by leoheck