kreso-t/cncjs-kt-ext #16: Extra diagonal lines being generated.

leoheck, 2020-01-18:
After making the auto-leveling, I see these diagonal lines appearing.
[image]

This is the original board loaded in CNCjs
[image]

Here is the end result in FlatCam.
[image]

I am making only the isolation routing.
I believe these lines are just the travel movement, but they are being drawn in my PCB as you can see below. 

**Do you have any idea why this happens?**

[image]

leoheck, 2020-01-18:
How can I export the altered file to compare against the original?

leoheck, 2020-01-18:
Ok, I found it. Disabling 3D-View on CNCjs we can download the file.

leoheck, 2020-01-18:
The main difference is that the #AL file has this `G00` and `G01` codes. I could fix (remove) the diagonal lines by commenting out the lines starting with `G00` and `G01` 

Here is the initial part of the file. 

My question is. These `G00` and `G01` command are right?
Is it possible that the auto leveler is adding this wrongly?

```
G21
G90
G94
F600.00
; G00 X0.000 Y0.000 Z0.019 ; Z0.000 <======== Commented
; G00 X0.000 Y0.000 Z2.019 ; Z2.000 <======== Commented
M03S20000
G4P1
G00x2.417y9.567
; G01 X0.000 Y0.000 Z2.019 ; Z2.000 <======== Commented
; G01 X0.000 Y0.000 Z-0.031 ; Z-0.050 <======== Commented
G01x2.412y9.517
G01x2.434y9.294
G01x2.499y9.080
G01x2.605y8.883
```

leoheck, 2020-01-18:
Here are a side by side comparison between my original gcode (left) and auto leveled gcode (right)

It looks like it is adding `G00 X0.000 Y0.000 Z-0.029 ; Z-0.050` at line `1064`. It doesn't look right since there is no command like this in the original file. 

[image]

Also, extra `X0.000 Y0.000` commands added doesn't look right either.

leoheck, 2020-01-19:
I am cleaning/fixing the auto leveled .gcode with my [FIXES.sh](https://www.dropbox.com/s/9djr69nkn2dd3e3/FIXES.sh?dl=0) script by now. (however, it will be better to fix it in the auto leveler)

The method I am using:
1. Upload the original.gcode to the CNCjs.
2. Perform the auto-level.
3. Download the autoleveled.gcode 
4. Fix it by running the script `./FIXES.sh autoleveled.gcode`
5. It will generate the file `autoleveled_fixed.gcode` which I am uploading to the CNCjs again.

kreso-t, 2020-01-20:
Hi @leoheck, 

Can you post your original gcode where you noticed this problem, I will try to fix it within the plugin based on your example as this did not happen with any of my PCBs so far.

BR

kreso-t, 2020-01-20:
Hi @leoheck ,

Based on the image you put in the post above the problem seem to be that your xy coordinates are in lower-case, and the plugin is case sensitive (and has been processing only upper-case XY letters). I have fixed that now, it should  work now also with lower-case xyz.

BR

leoheck, 2020-01-20:
Ah, I see. 
I am converting everything to uppercase, so the GCODE syntax highlight that I use on sublime text works. I am also adding spaces to make easier to read.

I have the files on my dropbox, here, they are.

1. Original GCODE from Flatcam
[board_test_increased_holes_rotate-B_Cu.gcode](https://www.dropbox.com/s/dgvq2lsikoimdix/board_test_increased_holes_rotate-B_Cu.gcode?dl=0)
2. Autoleveled GCODE (exported from CNCjs)
[board_test_increased_holes_rotate-B_Cu_autoleveled.gcode
](https://www.dropbox.com/s/9wjkr3khh8y0m1m/board_test_increased_holes_rotate-B_Cu_autoleveled.gcode?dl=0)
3. This is my fixed (and easy to read) GCODE
[board_test_increased_holes_rotate-B_Cu_autoleveled_fixed.gcode](
https://www.dropbox.com/s/cs7z6bxmhrs9ud8/board_test_increased_holes_rotate-B_Cu_autoleveled_fixed.gcode?dl=0)

kreso-t, 2020-01-21:
Autoleveler adds this "; Zxy" at the end i.e.
`G01 X0.000 Y0.000 Z-0.029 ; Z-0.050`
 
and  the lines that don't have that i.e.
`G01 x2.412 y9.517`
were actually skipped and only copied to output (they were not recognized due to lower case xy) and no leveling was applied for those lines

pls. try with the most recent version of the plugin, it should be ok also with lower case coordinates

leoheck, 2020-01-21:
Sweet, thanks. I am leaving on a vacation tomorrow. I hope to have time to test it by the end of the day today.

leoheck, 2020-01-22:
Now it looks interesting. The diagonal lines are gone. I confess that I was a bit disappointed when I saw that the auto-leveling was only a few commands before each. But now it really looks good. I am eager to give it a try, but I can do it only in some weeks =/ 

Thanks, dude!

```
G90
G94
F600.00
G00 X0.000 Y0.000 Z0.021 ; Z0.000
G00 X0.000 Y0.000 Z2.021 ; Z2.000
M03S20000
G4P1
G00 X0.000 Y0.000 Z2.021 ; Z2.000
G00 X1.225 Y4.848 Z2.012 ; Z2.000
G00 X2.417 Y9.567 Z2.001 ; Z2.000
G01 X2.417 Y9.567 Z2.001 ; Z2.000
G01 X2.417 Y9.567 Z-0.049 ; Z-0.050
G01 X2.417 Y9.567 Z-0.049 ; Z-0.050
G01 X2.412 Y9.517 Z-0.049 ; Z-0.050
G01 X2.412 Y9.517 Z-0.049 ; Z-0.050
G01 X2.434 Y9.294 Z-0.049 ; Z-0.050
G01 X2.434 Y9.294 Z-0.049 ; Z-0.050
G01 X2.499 Y9.080 Z-0.049 ; Z-0.050
G01 X2.499 Y9.080 Z-0.049 ; Z-0.050
G01 X2.604 Y8.882 Z-0.049 ; Z-0.050
G01 X2.604 Y8.882 Z-0.049 ; Z-0.050
G01 X2.746 Y8.709 Z-0.049 ; Z-0.050
G01 X2.746 Y8.709 Z-0.049 ; Z-0.050
G01 X2.783 Y8.680 Z-0.050 ; Z-0.050
G01 X2.783 Y8.680 Z-0.050 ; Z-0.050
G01 X2.796 Y8.667 Z-0.050 ; Z-0.050
G01 X2.796 Y8.667 Z-0.050 ; Z-0.050
G01 X4.754 Y6.709 Z-0.057 ; Z-0.050
G01 X4.754 Y6.709 Z-0.057 ; Z-0.050
G01 X4.767 Y6.695 Z-0.057 ; Z-0.050
G01 X4.767 Y6.695 Z-0.057 ; Z-0.050
G01 X4.796 Y6.659 Z-0.057 ; Z-0.050
G01 X4.796 Y6.659 Z-0.057 ; Z-0.050
G01 X4.969 Y6.517 Z-0.057 ; Z-0.050
G01 X4.969 Y6.517 Z-0.057 ; Z-0.050
G01 X5.167 Y6.412 Z-0.054 ; Z-0.050
G01 X5.167 Y6.412 Z-0.054 ; Z-0.050
G01 X5.381 Y6.347 Z-0.055 ; Z-0.050
G01 X5.381 Y6.347 Z-0.055 ; Z-0.050
G01 X5.604 Y6.325 Z-0.056 ; Z-0.050
G01 X5.604 Y6.325 Z-0.056 ; Z-0.050
G01 X5.651 Y6.329 Z-0.056 ; Z-0.050
G01 X5.651 Y6.329 Z-0.056 ; Z-0.050
G01 X5.669 Y6.330 Z-0.056 ; Z-0.050
G01 X5.669 Y6.330 Z-0.056 ; Z-0.050
```

leoheck, 2020-04-02:
It is working well. I will close this.

closed 2020-04-02 by leoheck