kreso-t/cncjs-kt-ext #19: Strip double quotes from secret

aroman12, 2020-03-15:
The currently latest version of cncjs (1.9.20) running on raspbian buster puts double quotes around the secret string within the ~/.cncrc file, so  the autolevel script fails with:

Connection error. { message: 'invalid signature',
  code: 'invalid_token',
  type: 'UnauthorizedError' }

The quotes and extra spaces should be stripped. I did it using 
export CNCJS_SECRET=$(grep secret ~/.cncrc | sed "s/[ \t]\+//g" | tr ',' ' ' |cut -d':' -f2 | sed 's/"//g' | sed 's/ //g')

Then it worked:
pi@marvin:~/cncjs-kt-ext $ node . --port /dev/ttyUSB0 
Connected to ws://localhost:8000?token=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpYXQiOjE1ODQyOTE1NzMsImV4cCI6MTU4Njg4MzU3M30.Cx8GRr6STsfxvJZA73OSEem2aOsLBnkwXdYzct1rJxE
Connected to port "/dev/ttyUSB0" (Baud rate: 115200)

By the way, this extension works great and is a must for anyone using cncjs to make PCBs. Thank you very much for providing it.

leoheck, 2020-03-19:
Interestingly, I am using the same version `1.9.20` I am the one that added that info to the README. But if this is working for you, I think it is good to update the README, since your command is removing the quote marks.

I totally agree with you. This is a really good plugin.
There is a thread on cncjs discussing to add this plugin to the core of the cncjs.

leoheck, 2020-03-20:
hi @kreso-t you can close this issue now.

kreso-t, 2020-03-20:
thanks
br

closed 2020-03-20 by kreso-t