kreso-t/cncjs-kt-ext #21: Reapply bed level in new gcodes.

leoheck, 2020-04-02:
Hi, I am loading many gcodes to make one board, and after some time I cannot use the autolevel command again on the same board since there will be areas without copper.

My question is when I load a new gcode, does the cncjs-kt-ext apply the previously generated configs in Z-axis in this new file? 

If not, is it possible to create a new command to update the be gcode with the later configuration? For example, it could be called something like "reapply_last_autolevel_config"

kreso-t, 2020-04-08:
Hi,

I added this option, you can use it by issuing command:
 (#autolevel_reapply)
need to update in readme still

Tnx for suggestion! This is a useful feature.

BR

leoheck, 2020-04-08:
Really?! I was going to look this right now. You are the man!

leoheck, 2020-04-08:
I just have a question before starting tests.

Most of the time the same trill kit is used, and the drill can have the same size when in the chuck.
But I personally have a misc drills brits. The end position inside the drill chuck can vary.

I would like to ask if this new feature is taking account of the new Z-offset to reapply the values.

For example, I will change the drill bit, then I can go to an area with copper to probe de Z-height again. Then I would like to #autolevel_reapply considering this new Z-offset.

leoheck, 2020-04-08:
I can also add it to README, I will make a PR.

kreso-t, 2020-04-09:
> I just have a question before starting tests.
>
> Most of the time the same trill kit is used, and the drill can have the same size when in the chuck.
> But I personally have a misc drills brits. The end [...]

Hi, for this after changing drill to another having different length, you can Z-Probe the PCB surface at approximately starting point (xmin,ymin) and set the Z WCO to zero again. This is what I usually do.

Another possibility may be to use G43.1 command (Tool length offset) - I did not try that so far

BR

kreso-t, 2020-04-09:
> I can also add it to README, I will make a PR.

Tnx!!

leoheck, 2020-04-09:
That is what I was looking for, thanks. I will add this in the readme as well.

> ... after changing drill to another having different length, you can Z-Probe the PCB surface at approximately starting point (xmin, ymin) and set the Z WCO to zero again. This is what I usually do.

closed 2020-05-08 by leoheck