kreso-t/cncjs-kt-ext #23: How to redirect the plugin output to the CNCJs console or ins a custom widget?

leoheck, 2020-04-09:
Right now, I am using the terminal output to see `cncjs-kt-ext` progress. I dont't like this, because I have to ssh to my RPi. So, do you know if there is a way to see its output somewhere embedded in the CNCJs page?

kreso-t, 2020-04-09:
You can see some most important output messages in the **Console** widget of the CNCjs page. We can add additional messages there if necessary, but what is currently logged there it seems enough to figure out if all is running OK.

leoheck, 2020-04-09:
I see. I was thinking to have at least some progress of the z-offset being probed, so we can visualize it is working and if the offsets are good enough (inside of a small delta)

leoheck, 2020-04-09:
Ah, maybe if at the end of the process, you log the min, max and mean offset value will be nice info to understand the quality of the process.

closed 2020-05-08 by leoheck