kreso-t/cncjs-kt-ext #28: Windows?

GeerGuy, 2020-09-14:
Can this be installed and used with the Windows versions of CNCjs?

If so, please provide install instructions.

M0USS, 2020-09-14:
Hello, is it possible to install kreso-t / cncjs-kt-ext on CNCjs APP Electron?
Thank you in advance.

kreso-t, 2020-09-15:
Hi,

It should be possible to use it with Windows and with Electron APP.  However I didn't try that yet (I am running it on Raspberry PI/Linux) , so I can't provide some specific instructions. Generally it should work in very similar way as on Linux+nodejs.

willmackey2nd, 2020-10-15:
Confirmed that it works on Windows. 
1. If you use the standalone application, you need to find out the port number. Open CNCjs View -> View In Browser and copy the port from the url, e.g. the 59539 part of http://127.0.0.1:59539/#/login

willmackey2nd, 2020-10-15:
Confirmed that it works on Windows. 

1. If you use the standalone application, you need to find out the port number. Open CNCjs View -> View In Browser and copy the port from the url, e.g. the 59539 part of http://127.0.0.1:59539/#/login
Note that this port number might/will change on every time you launch CNCjs app. IDK if there's a way to configure to use a static one.
2. Check from Device Manager which COM port was assigned for your CNC controller 
3. Launch auto leveler with command like:
`node . --port COM9 --socket-port 59539`
If it works it says the connection is ok, e.g. _"Connected to port "COM9" (Baud rate: 115200)"_
4. Check other possibly required arguments with node . --help.
For example if you have an CNCjs account set up you need to fill in the authentication stuff that you can find from C:\Users\<username>\.cncrc
In my case the whole launch command is like so
`node . --socket-port 59539 --id 'randomlettersandnumbersfrom.cncrc.file' --name 'MyAccountNamefron.cncrc.file' --secret 'morerandomlettersandnumbersfrom.cncrc.file' --port COM9`

closed 2021-03-18 by kreso-t