kreso-t/cncjs-kt-ext #31: After fail probe should return z to zero to avoid tool break.

diegomgarcia, 2020-11-05:
Hi guys this get me to broke two tools after probe failed, in other probes it usually return to zero.

I will take a look on how it can be done and do a PR. 

Any toughs on that ?

LongLiveCHIEF, 2021-03-18:
If the probe fails, how do you know where 0 is?

diegomgarcia, 2021-03-18:
Just return to the original position that was before start to probe.

LongLiveCHIEF, 2021-03-18:
I think the challenge is that if probing fails, the cncjs grbl client becomes blocked, and cncjs requires you use the client reset command in order to restore a grbl client connection.  This in turn means the cncjs-ket-ext needs to be restarted, and such loses everything it had in memory.

I think this feature request actually belongs in cncjs and not in this extension.

diegomgarcia, 2021-03-18:
I really didn't dig it in the code to see if this was the causa, I was thinking it could be handled something like that: 

start probe sent
store the current z position
begin probe
if fail return to stored z position
continue the normal process blocking the client ....

LongLiveCHIEF, 2021-03-20:
cncjs stops communicating with the mainboard on probe failure, so `if fail` isn't something that can be controlled by a plugin.

diegomgarcia, 2021-03-20:
> cncjs stops communicating with the mainboard on probe failure, so `if fail` isn't something that can be controlled by a plugin.

I have to disagree with that, the fail is just the fact that it didn't close the circuit, aka didn't received the touch point, so before it aborts the plugin is still processing it so why not just interpret as it failed and retract the z to the stored position ?

LongLiveCHIEF, 2021-03-20:
Look, there's a communication process that happens here for data flow between 3 systems; the fimrware of your machine <--> cncjs <--> autolevel plugin.

The each communicates with each other by sending commands, and getting responses. When responses are received, all is well. However, when responses _aren't_ received, communication is permanently disrupted. 

Autolevel _never_ communicates directly to the machine, it always goes through cncjs. cncjs then passes on the commands to move and probe each point, and then it fires a "point probed" event that autolevel listens and waits for, then tells cncjs to instruct the machine to probe another point. 

CNCjs _is_ aware of position when it sends the probe command to the firmware, but the firmware executes the command, and has it's own logic on what to do if the probe fails.  This is not something CNCjs can control.

In the event of a probe failure of the type you describe, it currently sends a response back that prevents CNCjs from being able to send any further commands over the existing communication pipeline. 

The only way currently to reestablish a new pipeline, is for CNCjs to reset the connection, at which point CNCjs loses/dumps positional information (or essentially sets the new position to 0).

When CNCjs loses the pipeline with the cnc machine, the autolevel plugin _also_ loses connection to CNCjs.  And has to reestablish a new connection to CNCjs by restarting... which dumps all map data held in memory.

You're not wrong about how things work electrically, but the problem isn't in the electronics, it's in the firmware and software components on the other end of the pipeline.

LongLiveCHIEF, 2021-03-20:
tl;dr, this is a feature request for cncjs, not for the autoleveler.  Once implemented in cncjs, the autoleveler would be able to look at what it does when it hears probe failure events from cncjs while its trying to create a map.

diegomgarcia, 2021-03-25:
Thanks @LongLiveCHIEF I didn't fully understand that this pipeline was ended by the firmware, I was thinking the plugin was calling the cncjs to end the communication when it received an failed event, but it seems that the plugin never receives the failed event in the way you described the event flow. 

Eventually I will take a look at the source code of cncjs to see if it's possible to add that feature with the current architecture.

Thanks again for the fully details.

closed 2021-03-31 by diegomgarcia