kreso-t/cncjs-kt-ext #34: I cannot run the auto level comando twice.

leoheck, 2020-12-30:
How do I discard current values to perform the auto-level again?

I am running this as a service, and I do' want to restart the revise every time, so but it happens that it is not running 2 times in a row. I am not sure if I am doing something wrong, or if this is something that can be improved. 

For instance, a command to discard current values would be a good idea if it doesn't have such a thing.

What do you think @kreso-t

kreso-t, 2021-03-18:
Sorry for late reply

If you run the auto-level again previous points should be discarded by the following line in the start() function:

`this.probedPoints = []`

and then new points measured, restart should not be needed. Unless there is a bug that happens in some very specific situation. 

@leoheck

leoheck, 2021-03-18:
Hey @kreso-t what's up? I was able to run the auto-level multiple times one after another and it was fine. But I think this issue was about another weird situation that I cannot remember anymore. Well, thanks for your answer, I am closing this now.

closed 2021-03-18 by leoheck