kreso-t/cncjs-kt-ext #45: nothing happens

jappyjan, 2022-02-04:
in the linux console i can verify that the extension is connected to the serial port as well as the websocket (at least the logs say so).
but when i send "(#autolevel)" through the ui console OR through a macro, nothing happens and i also do not get any further logs...

jkesseler, 2022-02-05:
From the readme:

> Please, note that this command will be **ignored when put inside the gcode file or type it in the console**, you must run it from a macro.

If I understand correctly you'll have to create a macro in CNCJs with the contents `(#autolevel)`, load some g-code and then run that macro.

jappyjan, 2022-02-05:
As said above: I tried both

IQworkshop, 2022-02-15:
double check the index.js in the install folder.
check that the 2 references to the tty device (ttyACM0, ttyUSB0 etc) is correct for your configuration

Also check in the same file that the port definition (2 references) matches the port CNCjs is running on (default is 8000)

I had to modify both of these to make the plugin work for me after having the same behavior you describe

jappyjan, 2022-02-15:
Thanks a lot! I will have a look and maybe even open a PR to load these from the cncjs config 👍🏼

closed 2022-07-26 by kreso-t