kreso-t/cncjs-kt-ext #52: Documentation problem

AuBeTeX, 2022-07-19:
Hi!

Sorry my english, and the bothering...

The docimentation is a bit incomprehensible. The probeopen needs to run before autolevel, or after autolevel? What happened if i use probeopen and the file is exists? The exists file is applied to gcode, or the exists file overtwritten by the new autolevel datas? 

In my opinion this needs to clear. Maybe need a probesave command, or a overwrite parameter to the probeopen command. 

What if the gcode has an applied autolevel. the new autolevel owervrites that or modify the previous autolevel datas?

How to stop the autoleveling?
 The autoleveler can work in:
(probeopen)
(#autolevel)
(probeclose)

the autolevel reapply working from previously saved file? How? need a probeopen before?

AuBeTeX, 2022-07-20:
Hi again!

I'm trying and trying, but don't understand what doing wrong.
This is the result after a 2.5mm autolevel with 0.1 mm tool 2 pass 10% overlap 2 level multi depth (0.02, 0.04). The result is horrible. Some point the tool tore it apart the PCB some point barely reaches it. 
[image]

kreso-t, 2022-07-20:
can you describe what steps are you doing and what output you get from CNCjs?
Do you get after  (#autolevel) is completed that gcode loaded in CNCjs file was modified i.e. Z coordinates are altered?

AuBeTeX, 2022-07-21:
Yes, the Z coordinates modified. The save function can't use. if i use before the auto level the file is 0 byte long, after use 0 to. Somehow in one time, i'm succesfully saved the data, but never again. 

I try this macro:
```
; Z-Probe
G00 X0 Y0
G91
G38.2 Z-40 F100
G10 L20 P1 Z0
G0 Z0.5
G38.2 Z-10 F10
G90
; Set the active WCS Z0
G10 L20 P1 Z0
; Retract from the touch plate
G91
G0 Z2
G90
M30
(probeopen heightmap.map)
(#autolevel D2 H2.0 F20 M0.2)
(probeclose)
```

later i'm deleted the probeopen, probeclose commands.
The screen i see the "AL: xxx.grbl" text. And see the altered grbl. 

the console output on end of the autoleveling:
```
[PRB:-18.960,-66.200,-36.780:1]
ok
feeder> (AL: probing point 522)
ok
feeder> G90 G0 X57.070 Y-0.200 Z2
ok
feeder> G38.2 Z-3 F20
ok
feeder> G0 Z2
[PRB:-16.930,-66.200,-36.780:1]
ok
ok
feeder> (AL: dz_min=-0.052, dz_max=0.738, dz_avg=0.319)
ok
feeder> (AL: applying compensation ...)
feeder> (AL: loading new gcode #AL:teszt-B_Cu.gbr_iso_combined_cnc.nc ...)
ok
ok
feeder> (AL: finished)
```

the modified grbl after the autolevel:

```
G21
2G90
3G94
4G01F120.00
5M5
6G00 X0.000 Y0.000 Z15.082 ; Z15.000
7G00 X0.000 Y0.000 Z15.082 ; Z15.000
8G00 X0.000 Y0.000 Z15.082 ; Z15.000
9T1
10M6
11M0
12G00 X0.000 Y0.000 Z15.082 ; Z15.000
13G00 X0.000 Y0.000 Z15.082 ; Z15.000
14M03S200.0
15G01F120.00
16G00 X0.000 Y0.000 Z15.082 ; Z15.000
17G00 X6.100 Y-13.915 Z15.053 ; Z15.000
18G01F60.00
19G01 X6.100 Y-13.915 Z15.053 ; Z15.000
20G01 X6.100 Y-13.915 Z0.036 ; Z-0.018
21G01F120.00
22G01 X6.100 Y-13.915 Z0.036 ; Z-0.018
23G01 X6.100 Y-14.216 Z0.035 ; Z-0.018
24G01 X6.100 Y-14.216 Z0.035 ; Z-0.018
25G01 X6.101 Y-14.235 Z0.035 ; Z-0.018
26G01 X6.101 Y-14.235 Z0.035 ; Z-0.018
27G01 X6.103 Y-14.252 Z0.035 ; Z-0.018
28G01 X6.103 Y-14.252 Z0.035 ; Z-0.018G21
2G90
3G94
4G01F120.00
5M5
6G00 X0.000 Y0.000 Z15.082 ; Z15.000
7G00 X0.000 Y0.000 Z15.082 ; Z15.000
8G00 X0.000 Y0.000 Z15.082 ; Z15.000
9T1
10M6
11M0
12G00 X0.000 Y0.000 Z15.082 ; Z15.000
13G00 X0.000 Y0.000 Z15.082 ; Z15.000
14M03S200.0
15G01F120.00
16G00 X0.000 Y0.000 Z15.082 ; Z15.000
17G00 X6.100 Y-13.915 Z15.053 ; Z15.000
18G01F60.00
19G01 X6.100 Y-13.915 Z15.053 ; Z15.000
20G01 X6.100 Y-13.915 Z0.036 ; Z-0.018
21G01F120.00
22G01 X6.100 Y-13.915 Z0.036 ; Z-0.018
23G01 X6.100 Y-14.216 Z0.035 ; Z-0.018
24G01 X6.100 Y-14.216 Z0.035 ; Z-0.018
25G01 X6.101 Y-14.235 Z0.035 ; Z-0.018
26G01 X6.101 Y-14.235 Z0.035 ; Z-0.018
27G01 X6.103 Y-14.252 Z0.035 ; Z-0.018
28G01 X6.103 Y-14.252 Z0.035 ; Z-0.018
```

I would note it: All of comments is gone. 

I not sure what make the mistake. Maybe z axis inaccurate? (the cnc is in my closet almost 2 years) slow controller board? (the modern board burned down due to an inaccurate heightmap -Candle- i'm don't understand the protectionless electronics ) 

If  i can suggest a speedup trick...

on scanning i give a smaller height if this height relative, to last touch point so i give 0.5mm to scan height, if this 0.5 counted from the last touched height. the touch rarely change 0.5mm in 10mm distance, but the scanning speed is significant faster. 

Thanks to the asistance, and your work

AuBeTeX, 2022-07-21:
Hi!

i'm testing a little with touch probe... if i'm used a higher feed rate, the result is inaccurate. 

F20 +/- 0.2 mm
F5 +/- 0.09mm
F1 +/- 0.003mm

i created an own probe macro. with this macro the touching error always betveen +/- 0.003mm to 0.006mm (by the cncjs report)

```
G91
G38.2 Z-10 F5
G10 L20 P1 Z0
G0 Z0.2
G38.2 Z-10 F1

```

 maybe an another suggestion. the autoleveler probe's it could be like this:

```
G91
G38.2 Z-10 F<X>
G10 L20 P1 Z0
G0 Z0.<X>
G38.2 Z-0.(<X>+0.1) F1

```
This is may create a better result. i try an autoleveling some specific feed rate. and (if) i can save the result compared to this. 

of course this problem maybe exists only in hobby cnc machines.

kreso-t, 2022-07-21:
It looks to me like you are doing everything correctly regarding the software use. 

It could be then that the problem is mechanical. If you do get good probing repeatability then it could be that machine is accurate when there are no forces involved, but not so much when it as actually milling. Also for PCB milling, backlash should be minimal (i.e. less than 50 um) in order to get good results - either by using ball screws or some sort of anti-backlash nut

AuBeTeX, 2022-07-21:
ok definitely hardware problem. If your code has a little bit more comment, maybe i'm understand it. I'm not programmed js before. But if i see right, i need an average auto leveling. minimum 3 but preferably 5 touch average. Then  maybe i can modify the leveler. Until that i trying. 

Thank you very mutch!

AuBeTeX, 2022-07-22:
Sorry again! 

i found a problem. Your code not analyze the movement if i see it right. I created a 20X20mm square, with isolation routing, so this is two square. an inner and an outer. The flatcam generate a wrong code, that an anoter bug report. The outer square has 4 movement.  X19.9Y-19.9 then move X19.9Y-0.1 then X0.1X-19.9 then X0.1Y-0.1 then X0.1Y-19.9. your code that modified this 4 lines. But the heightmap shows differences in height in this 20 mm. Normally this movement needs to chop multiple pieces. Minimum so mutch as the heightmap probes in that distance. Here has one 20mm step, but i have 10 pieces 2mm height probe. So this movement needs to divide to 10 parts, then apply the Z axis movement. 

So unfortunately my hardwer is not the best on market, but this problem can cause my problems one part. 

What you think?

This is the modified grbl as you see the flatcam make a mistake too. The inner square is 4 movement the outer is 67. That is not a square maybe a diamond? i'm not created a visual image. But the modified GRBL has some duplicates. 

```
G21
G90
G94
G01F20.00
M5
G00 X0.000 Y0.000 Z14.794 ; Z15.000
G00 X0.000 Y0.000 Z14.794 ; Z15.000
G00 X0.000 Y0.000 Z14.794 ; Z15.000
T1
M6
M0
G00 X0.000 Y0.000 Z14.794 ; Z15.000
G00 X0.000 Y0.000 Z14.794 ; Z15.000
M03S200.0
G01F20.00
G00 X0.000 Y0.000 Z14.794 ; Z15.000
G00 X0.100 Y-19.900 Z15.227 ; Z15.000
G01F60.00
G01 X0.100 Y-19.900 Z15.227 ; Z15.000
G01 X0.100 Y-19.900 Z0.177 ; Z-0.050                 //Down Z
G01F20.00
; Inner square
G01 X0.100 Y-19.900 Z0.177 ; Z-0.050               //Down again?
G01 X19.900 Y-19.900 Z0.111 ; Z-0.050             //move to first corner
G01 X19.900 Y-19.900 Z0.111 ; Z-0.050            //move again?    its new
G01 X19.900 Y-0.100 Z-0.137 ; Z-0.050            //move to second corner
G01 X19.900 Y-0.100 Z-0.137 ; Z-0.050            //move again?    its new
G01 X0.100 Y-0.100 Z-0.255 ; Z-0.050              // move third corner
G01 X0.100 Y-0.100 Z-0.255 ; Z-0.050              //move again?  its new
G01 X0.100 Y-19.900 Z0.177 ; Z-0.050             //move to fourth corner
G00 X0.100 Y-19.900 Z0.177 ; Z-0.050             // move again.... Its new

G00 X0.100 Y-19.900 Z2.227 ; Z2.000
G00 X0.100 Y-19.900 Z2.227 ; Z2.000
G00 X-0.100 Y-20.003 Z2.245 ; Z2.000

G01F60.00
G01 X-0.100 Y-20.003 Z2.245 ; Z2.000
G01 X-0.100 Y-20.003 Z0.195 ; Z-0.050
G01F20.00
G01 X-0.100 Y-20.003 Z0.195 ; Z-0.050
G01 X-0.100 Y-20.010 Z0.196 ; Z-0.050
G01 X-0.100 Y-20.010 Z0.196 ; Z-0.050
G01 X-0.098 Y-20.020 Z0.197 ; Z-0.050
G01 X-0.098 Y-20.020 Z0.197 ; Z-0.050
G01 X-0.096 Y-20.029 Z0.199 ; Z-0.050
G01 X-0.096 Y-20.029 Z0.199 ; Z-0.050
G01 X-0.092 Y-20.038 Z0.200 ; Z-0.050
G01 X-0.092 Y-20.038 Z0.200 ; Z-0.050
G01 X-0.088 Y-20.047 Z0.202 ; Z-0.050
G01 X-0.088 Y-20.047 Z0.202 ; Z-0.050
G01 X-0.083 Y-20.056 Z0.203 ; Z-0.050
G01 X-0.083 Y-20.056 Z0.203 ; Z-0.050
G01 X-0.077 Y-20.064 Z0.204 ; Z-0.050
G01 X-0.077 Y-20.064 Z0.204 ; Z-0.050
G01 X-0.071 Y-20.071 Z0.205 ; Z-0.050
G01 X-0.071 Y-20.071 Z0.205 ; Z-0.050
G01 X-0.064 Y-20.077 Z0.206 ; Z-0.050
G01 X-0.064 Y-20.077 Z0.206 ; Z-0.050
[... 198 lines cut ...]
G01 X-0.0098 Y0.0995
G01 X-0.0195 Y0.0981
G01 X-0.0290 Y0.0957
G01 X-0.0383 Y0.0924
G01 X-0.0472 Y0.0882
G01 X-0.0556 Y0.0832
G01 X-0.0635 Y0.0773
G01 X-0.0707 Y0.0707
G01 X-0.0773 Y0.0635
G01 X-0.0832 Y0.0556
G01 X-0.0882 Y0.0472
G01 X-0.0924 Y0.0383
G01 X-0.0957 Y0.0290
G01 X-0.0981 Y0.0195
G01 X-0.0995 Y0.0098
G01 X-0.1000 Y-0.0000
G01 X-0.1000 Y-20.0025
G00 Z2.0000
M05
G00 Z2.0000
G00 Z15.00

```

I need that PCB for now so i'm search my iron-chlorid and the laser printer, thats enough to an one sided PCB

kreso-t, 2022-07-22:
this funcion splitToSegments(p1, p2, units) does the chopping 

https://github.com/kreso-t/cncjs-kt-ext/blob/e5bb26d4cbf9cc7d47223f3674f23dc50cb31da4/autolevel.js#L343

so far it all worked fine for me 

i.e. this is the last PCB I milled few months ago, the size is quite large 100x150 mm and it came out ok:
[image]

also I usually plot the silkscreen layer with the white marker for glass (POSCA 0.7mm) mounted on the CNC  (though have to set larger fonts for silkscreen layer and adjust it a little because pen is not thin enough for fine details)
[image]

AuBeTeX, 2022-07-23:
Hi! Nice work, what programs are you use? i created the PCB in Kicad then convert in Flatcam to grbl then use cncjs. I use the candle before but that always dropped the connection.

And what you think, in my case why not segmentated the movement?

AuBeTeX, 2022-07-23:
An another question. The autoleveler set the initial Z height, or use the current z settings? So if i set the Z lower a few mikrometer... i se the first is not the best, but i'm not create a new grbl to deeper cut only i set the z to a bit lower (G10 L20 P1 Z-0.05)... Can work?

kreso-t, 2022-07-24:
> Hi! Nice work, what programs are you use? i created the PCB in Kicad then convert in Flatcam to grbl then use cncjs. I use the candle before but that always dropped the connection.
> 
> And what you think, in my case why not segmentated the movement?

thx, I also use kicad + flatcam + grbl + cncjs + a small script  to generate grbl gcode from gerber file for silkscreeen plotting.

however **I am using some very old version of this autolevel script**, I was accepting the pull requests in order for other people to use new features, but I didn't upgrade on the CNC machine it for a long time because the original version worked just ok for me .

But now I see that in this current version it may be a problem here 

https://github.com/kreso-t/cncjs-kt-ext/blob/e5bb26d4cbf9cc7d47223f3674f23dc50cb31da4/autolevel.js#L471

it looks like it is missing the units parameter ...  So I will need to test this new version and correct it

kreso-t, 2022-07-24:
> An another question. The autoleveler set the initial Z height, or use the current z settings? So if i set the Z lower a few mikrometer... i se the first is not the best, but i'm not create a new grbl to deeper cut only i set the z to a bit lower (G10 L20 P1 Z-0.05)... Can work?

This should work after autoleveling is performed - you can adjust machine work offset to get deeper cuts

kreso-t, 2022-07-24:
Hi,

yes, after testing I see that the splitToSegments didn't seem to work correctly in the recent versions.  Thanks for noticing this!

It is fixed now, here is the output example with new version using your gcode file:

[#AL gcode-testfile.txt](https://github.com/kreso-t/cncjs-kt-ext/files/9175956/AL.gcode-testfile.txt)

hopefully this also solves the problem for your PCB

AuBeTeX, 2022-07-25:
Hi!

Thank you! I'm testing.

AuBeTeX, 2022-07-25:
Hi!

It's perfect. The PCB is almost perfect in first try!

Thank you!

closed 2022-07-26 by kreso-t