kreso-t/cncjs-kt-ext #54: Problem in big files, and modificate in grbl files.

AuBeTeX, 2022-07-28:
Hi!

Sorry about this new bothering. I'm created a copper removal GRBL. The file (because the 8.99 flatcam is buggy, and only one thin tool is used) is a little big (33mb). Unfortunately, i'm deleted after unsuccessful auto leveling. The autoleveler, is got the file and the logs shows the working on it. 0-to ~150000 lines. Then Printed in the log the reloading message, but the modified file is not loaded to cncjs. It is not a big problem, the file is big because the flatcam is do something wrong. i'm cant reproduce this big file. 

Now created a smaller multi tool GRBL on this, the work is done. But looks like something not ok. See the attached picture. The isolation routing dreamy nice and precise, but the copper cleaning is odd. I'm attached the original grbl if you testing it. Im attached the height map, but i'm can't known where is the uploaded or modified GRBL in cncjs,

And another problem, when the program is modify the GRBL why delete the notes? When comes a toolchange routine, i not see what tool is need, because the tool dia message is vanished. 
"
M6    
(MSG, Change to Tool Dia = 1.9600)
M0
"
in a two tool GRBL is not a big problem, but in a 10 tool GRBL is mutch bigger. 

Thank you for work!

[heightmap.txt](https://github.com/kreso-t/cncjs-kt-ext/files/9207325/heightmap.txt)
[cnc_7z.zip](https://github.com/kreso-t/cncjs-kt-ext/files/9207351/cnc_7z.zip)
[image]

kreso-t, 2022-07-31:
Hi, 

About the big files - I noticed this before,  the problem seem to be in CNCjs socket API, upload of big files to CNCjs via socket API seem to fail (above certain file size).  Maybe there is a setting in CNCjs to remove this limit - to be checked if this becomes a problem for normal use. There is also an option (-o) to define the output directory where the output auto-leveled file is stored - so if the automatic upload to CNCjs fails, you can always manually load the output file into CNCjs.

As for copper cleaning paths - I did not find the copper cleaning gcode file in the attached zip to check what may be the problem:

[image]

About removing the comments/notes - I created a fix so that full line comments are preserved i.e. comment like
(MSG, Change to Tool Dia = 1.9600)
would be simply copied to output.

BR,
Kresimir

AuBeTeX, 2022-08-03:
Hi!

Yes the cleaning file is left out. in the attachment has 2. With multi tool and a one tool version The multi tool version is looks like an one tool but no. Flatcam error. 

Thanks the fix.
[cnc2.zip](https://github.com/kreso-t/cncjs-kt-ext/files/9250513/cnc2.zip)

AuBeTeX, 2022-08-07:
Hi!

A question... While your program modify the cnc file, put someone a modified grbl? So the program do that, i can download, and upload manulally to cncjs avoided the cncjs socket api problem.

I created an 5 pass 2 level isolation routing, then (of course) the AL is not succes that file :)

Thanks Again!

kreso-t, 2022-08-08:
Hi,

There is an -o <path_to_output_folder> command line option --> if this option is set to some existing folder it will put there modified gcode file. 

BR

AuBeTeX, 2022-08-08:
Sorry, if i'm stupid...

This command line option is where? 

This is not working:

```
cd /root/cncjs-kt-ext/
node . --socket-port 80 --port /dev/ttyUSB0 -o /home/root/
```

kreso-t, 2022-08-09:
Hi, 

this works for me:

`bash
/usr/local/bin/node /home/pi/cncjs-kt-ext/index.js -o /home/pi/cncprog
`
maybe you have no write rights to "/home/root/" or that folder does not exist (did you mean actually to use /root/)?

Note also that the file is created there **after** whole process is completed.

BR,
Kresimir

AuBeTeX, 2022-08-09:
Yes, i'm run all with root to be avoided all permission problem. So this command need to run after applying AL in the console?

I'm Trying.

Thanks!

kreso-t, 2022-08-09:
> So this command need to run after applying AL in the console?

No, you need to start the cncjs-kt-ext with that option -o before using AL (and keep it running, as you would normally use it). The program will write the file to that defined output folder after AL is completed and not during AL is being executed.

AuBeTeX, 2023-01-02:
Hi!

Sorry but i'm bother again. 

After a long pause, i'm try to create a new board. I reclone the kt-ext from github, and try to re apply the height map, that previously saved. But "no previous autolevel points" i try to open heightmap then close and overwrite the saved height map but no success. 

I'm can't create a new height map, because the PCB milling is started and always has some point on the milled route. 

How can i resolve this?

Thanks!

AuBeTeX, 2023-01-14:
Hi!

I have a question. Witch point is the zero? Where measure the initial point, the origin of autoleveling?

Thanks!

kreso-t, 2023-01-16:
Hi,

The lower left corner (xmin, ymin) is where the Z0 is set  --> this is the first point that is probed

i.e.
```
code.push(`G0 X${xmin.toFixed(3)} Y${ymin.toFixed(3)} Z${this.height}`)
code.push(`G38.2 Z-${this.height + 1} F${this.feed / 2}`)
code.push(`G10 L20 P1 Z0`) // set the z zero
```
BR,
Kresimir

AuBeTeX, 2023-01-16:
Hi!

Thanks the reply. But then i don't understand. I'm loaded a 0.3 mm circular tool, then set the zero. I'm loaded the copper cleaning grbl, started the autoleveler, with heightmap saving. When the autoleveler is done, i'm load the isolation map, apply the height map, then start milling. The milling is perfect. After that, i change the tool a 0.8 circular, then set the zero on x0,y0. After loaded the copper cleaning map, i'm applyed the height map, and start the milling. But this milling is horrible. Why working on isolation milling, and why not on copper cleaning? 
I'm Attached the files. and the picture.

Have any idea, what i'm doing wrong?
[image]

[230116.zip](https://github.com/kreso-t/cncjs-kt-ext/files/10424929/230116.zip)

kreso-t, 2023-01-16:
Can you try to zero second tool on  lower left corner ==> xmin, ymin (i.e. X4.77 Y-51.3) , 
It seems x0,y0 is actually upper left corner for your pcb, based files provided

AuBeTeX, 2023-01-16:
Hi!

Sorry, but something not clear. My board (the ncc) lover left corner is is x-0.6 y-51.7. The heightmap is created from ncc file.

So use that coordinates?

kreso-t, 2023-01-16:
It should be OK to zero it anywhere around that point (i.e. +-3mm), it does not have to be 100% exactly the same point as the height does not change significantly

AuBeTeX, 2023-09-27:
Hi!

I'm again... Sorry.

So I'm upgraded the cncjs to 1.10.3 and tryed to use. but the extension works odd. The scanning is done, The height map file looks good, The appliing is done look like without error But the result is:

32M03S300.0
33G01F60.00
34G00 X0.000 Y0.000 Z25.164 ; Z25.000
35G00 X1.166 Y-2.211 Z25.139 ; Z25.000
36G00 X2.333 Y-4.423 Z25.114 ; Z25.000
37G00 X3.499 Y-6.634 Z25.091 ; Z25.000
38G00 X4.665 Y-8.845 Z25.069 ; Z25.000
39G00 X5.832 Y-11.056 Z25.050 ; Z25.000
40G00 X6.998 Y-13.268 Z25.032 ; Z25.000
41G00 X8.164 Y-15.479 Z25.013 ; Z25.000
42G00 X8.481 Y-16.079 Z25.009 ; Z25.000
43G01F60.00
44G01 X8.481 Y-16.079 Z25.009 ; Z25.000
45G01 X8.481 Y-16.079 Z22.509 ; Z22.500
46G01 X8.481 Y-16.079 Z20.009 ; Z20.000
47G01 X8.481 Y-16.079 Z17.509 ; Z17.500
48G01 X8.481 Y-16.079 Z15.009 ; Z15.000
49G01 X8.481 Y-16.079 Z12.509 ; Z12.500
50G01 X8.481 Y-16.079 Z10.009 ; Z10.000
G01 X8.481 Y-16.079 Z7.509 ; Z7.500
52G01 X8.481 Y-16.079 Z5.009 ; Z5.000
53G01 X8.481 Y-16.079 Z2.509 ; Z2.500
54G01 X8.481 Y-16.079 Z0.009 ; Z0.000
55G01F60.00
56G01 X8.481 Y-16.079 Z0.009 ; Z0.000
57G01 X8.494 Y-16.168 Z0.008 ; Z0.000
58G01 X8.494 Y-16.168 Z0.008 ; Z0.000
59G01 X8.518 Y-16.262 Z0.007 ; Z0.000
60G01 X8.518 Y-16.262 Z0.007 ; Z0.000
61G01 X8.550 Y-16.354 Z0.007 ; Z0.000
62G01 X8.550 Y-16.354 Z0.007 ; Z0.000
63G01 X8.589 Y-16.436 Z0.006 ; Z0.000
64G01 X8.589 Y-16.436 Z0.006 ; Z0.000
65G01 X8.595 Y-16.448 Z0.006 ; Z0.000
66G01 X8.595 Y-16.448 Z0.006 ; Z0.000
67G01 X8.638 Y-16.519 Z0.005 ; Z0.000
68G01 X8.638 Y-16.519 Z0.005 ; Z0.000
69G01 X8.646 Y-16.531 Z0.005 ; Z0.000
70G01 X8.646 Y-16.531 Z0.005 ; Z0.000
71G01 X8.700 Y-16.603 Z0.005 ; Z0.000
72G01 X8.700 Y-16.603 Z0.005 ; Z0.000
73G01 X8.765 Y-16.675 Z0.004 ; Z0.000
74G01 X8.765 Y-16.675 Z0.004 ; Z0.000
75G01 X8.837 Y-16.740 Z0.004 ; Z0.000
76G01 X8.837 Y-16.740 Z0.004 ; Z0.000
77G01 X8.915 Y-16.798 Z0.003 ; Z0.000
78G01 X8.915 Y-16.798 Z0.003 ; Z0.000
79G01 X8.998 Y-16.848 Z0.003 ; Z0.000
80G01 X8.998 Y-16.848 Z0.003 ; Z0.000
81G01 X9.080 Y-16.887 Z0.002 ; Z0.000

there is no god for the PCB to be so straight, at least based on manual touches
So any idea? i know, i'm doing wrong :)

On the attachment is a scanned height map,  modified cnc file and the unmodified cnc file..

[problem-230927.zip](https://github.com/kreso-t/cncjs-kt-ext/files/12737261/problem-230927.zip)

AuBeTeX, 2023-10-10:
Hi!

Ok i'm doing wrong. Sorry.

A have another problem. But thats a next issue. 

Sorry again.

closed 2024-09-19 by AuBeTeX