kreso-t/cncjs-kt-ext #63: First point's probe height undesired offset

AndreaSpinetti, 2023-07-31:
I found that somehow the first point start measuring from 1mm above the 0, while the others from 1mm above the previous position (or absolute, not sure).
So if you are not manually probing in the same spot as the first auto level point you can have an undesired offset.

I found it by manually positioning the bit very close to the pcb and then zero the Z axis, after that I ran auto level as usual. Doing that I had always a pretty deep point in the area around the first autoprobe point, breaking few bits while trying to figure it out.

kreso-t, 2023-08-18:
I assume you are talking about reapplying the same autolevel measurement to a new toolpath? When you are reapplying the autolevel to a new toolpath/gcode, z offsets to that gcode are applied relative to first probed point (i.e. xmin, ymin) - for that point applied offset is ~0. So you need to zero the tool on that same point (or close to it) before starting the program.

AndreaSpinetti, 2023-08-18:
nope, just doing the autolevel sequence and then milling.
The problem affect just the first point, all the other are perfect.
Just try to zero Z at, for example 2mm from pcb, and start the autolevel, you will find the first point 2mm deeper than the other ± the measured offset.

By the way, the problem is easily solved adding a probe Z in the macro before autolevel, but you still have to go manually at the first scanning point location for better accuracy, which is not always (0,0).
[no problem at all, of course, just not possible doing a macro for that]

(also I have a question about the reapply, in which coordinate system is referred Machine or Work?)

sorynl, 2024-04-22:
> By the way, the problem is easily solved adding a probe Z in the macro before autolevel, but you still have to go manually at the first scanning point location for better accuracy, which is not always (0,0).

Do you mind expanding on that? I do have a z-probe script before Autolevel and still run in to this issue quite often. Not from reapplying but from the first scan. Only solution I have so far is to restart CNCjs and reload Autolevel.

delaneyparker, 2024-06-04:
I'm seeing the same thing. The first plunge the autolevel does drove the bit right through the PCB... pretty exciting 🤣

I tried again with the bit high above the work surface and brought the probe contact to the bit, and as @AndreaSpinetti said, the first point seems to plunge further than it should (and it also seems to ignore the probe input as it's plunging). After that, all the subsequent probes are nice and slow and register contact as expected.

I modified my macro to do a probe z before running autolevel but the issue is still happening; perhaps I need to modify my probe script to back off on Z even more before starting autolevel?

Here is the macro that I tried (and should reproduce the issue):
```
; Z-Probe
G91
G38.2 Z-10 F20
G90

; Set the active WCS Z0
G10 L20 P1 Z10

; Retract from the touch plate
G91
G0 Z4
G90

(#autolevel)
```

sorynl, 2024-06-19:
> Here is an example macro that resolved the issue for me:
> 
> ```
> G90 G21 G38.2 Z-10 F20
> G92 Z0
> G0 Z15
> M30
> (#autolevel)
> ```

Thank you for sharing, so far your solution worked flawlessly.

closed 2024-09-19 by kreso-t