kreso-t/cncjs-kt-ext #64: Autoleveling area problem

AuBeTeX, 2023-10-10:
Hi!

I'm bothering again. I known i'm boring.

I'm stingy. In my country the current PCB size is mostly 160X100mm If i mill a PCB and the pcb is smaller than size i'm reuse the pcb. Now i try to mill two pcb and i created two draw next to each other, and mill one after another. Less toolchange, less configuration. The first is no problem, the leveling is good, but the second's start and end point is X110Y0 to X145Y-45 If i start the autolevel, the autoleveling process not start the X110Y-45 but X0Y-45. But that area is milled, so, the AL is failed. Any idea? I know that create one cncjob with two project. But thets is not always possibly.

AuBeTeX, 2023-10-17:
Hi!

Sorry the late response. 

So i attached the asked files. But i created an all in one file too. I'm use that to autolevel, but somehow, after multiple tries, something wrong. I think if my machine is a piece of sht, but thats is the problem? i known the  made in "the third junk to left" chinese 10 cent price cnc is not the best tool, but a more people use that. If the precision is the problem, the errors are randomly. But see this:

[image]

this is the all in one pcb milled three times one after another. The milling speed is very slow, protect the tool, so the tool is undamaged. The upper area is on one line is deep. not a little. The other areas like a PCB petting zoo. The milling is -0.05 Z so if the little unprecision problem one or two 0.05 lower milling is solves it. But that upper Mariana trench? That almost a cut out depth.

Curiosity: If iy used the cncjob before the all in one, the AL and the milling is almost perfect. This is two example. But if the size growing something going wrong. See this:

[image]

Some idea? 
[all-in-one.zip](https://github.com/kreso-t/cncjs-kt-ext/files/12927151/all-in-one.zip)
[separate.zip](https://github.com/kreso-t/cncjs-kt-ext/files/12927152/separate.zip)

kreso-t, 2023-10-17:
what comes to my mind is maybe with larger PCB the problem is with the fixture i.e. clamping of the PCB to machine table. If clamped on the corners only, a larger PCB is more likely to flex under the tool (or even the probe during autolevel therefore lowering probing accuracy). For larger PCBs I sometimes use double-sided tape to attach it to the baseplate so that whole  PCB area is glued to the rigid surface

AuBeTeX, 2023-10-17:
Hi!

I'm not using clamping. i'm use full area double sided tape. All of the pcb-s same size 160X100mm only the used area is growing not the PCB. i see in the z-probe file the differences between +-0.7mm if i don't miss anithing. The clamping is not worked for me, because the problems you say. i'm trying now the double origin milling. I hope I'll be lucky

closed 2024-09-19 by AuBeTeX